pkerspe/ESP-FlexyStepper #2: Multi axis performance

vanniaz, 2020-08-07:
Hello, this is just a request for clarification, not a real issue.

I have to choose a library to control up to 6 stepper motors at the same time. I do not need interpolation, I only need to start the motors in parallel, possibly with different speeds and different target distances. The acceleration/deceleration ramps will be fixed and equal for all motors.
Now I see that your library is based on the original FlexyStepper, which has a poor performance on the AVR (7 kHz max for a SINGLE motor, while grbl on AVR is capable of 30 kHz with interpolation!).

You do not give any performance figure for your library on the ESP32, do you expect it to be capable of handling 6 axes in parallel? At what maximum frequency?

Of course I could try it but this takes some time to write a test program, checking the output with an oscilloscope etc., so maybe that you already have the answer.

Thank you very much.
Giovanni

pkerspe, 2020-08-07:
I cannot provide any performance measurements, the step signal generation is exactely the same as the one in flexystepper.
The processMovement function is called in a separate task that runs on the second core of the ESP32.

Performance is dependent on multiple factors:
- the interface you want to use to send new position settings to the ESP (obviously the webinterface is using more processing power than the serial Interfaxe)
- the Configured CPU speed of the ESP32
- the CPU time of you individual Code in the Loop function (if any)

Setting up the ESP Stepper Motor Server takes no longer than 5 minutes if you simply use the provided examples and use PlatformIO (if you have set it up before to have the required ESP32 libraries in place).

pkerspe, 2020-08-07:
BTW the original FlexyStepper was never tuned to performance as in maximum frequency but rather in smooth acceleration and the possibility to change target positions during movement. You should probably check out the SpeedyStepper library instead as stated in the readme of FlexyStepper

vanniaz, 2020-08-07:
Thank you for your reply.
Yes, I already have PlatformIO and the ESP32 libraries on my PC, so I will give it a try (I have to choose between ESP-FlexyStepper and Grbl_Esp32).
In my application a simultaneous start of all the axes is not required (i.e. a delay of even a few milliseconds would be tolerated).

closed 2020-08-07 by vanniaz