pkerspe/ESP-FlexyStepper #3: mm or degrees?

forna91, 2020-10-14:
Any solution for change the unit from mm to degrees? or reset the position when position is 360.000mm?

pkerspe, 2020-10-14:
You can just use steps if you want to set the shaft to a specific degree.
Usually most steppers have a step angle of 1.8 degree for a full step.
If you use microstepping you can break it down accordingly.
So it depends in general on your setup how many degrees of rotation is one step.
Does your stepper driver support microstepping?

In general you will have a Hard Time to move by full degrees if your stepper motor has 1.8 degree steps, since you would need to use a microstepping multiplier that allows you to get a small enough fraction of 1.8 to perform exactely 1 degree of rotation.
E.g. if you use 64 microsteps, the smallest rotation with one microstep is 1.8/64 = 0,028125 degrees.
Now the closest you get to 1 degree of rotation is by performing 35 or 36 microsteps, which will give you either 0,984375 degrees or 1,0125 degrees of motion, but never exactely one degree.

So you have two options:
1.) Increase the microstepping to the highest value your driver supports and get as close as possible to the required resolution you need
2.) use gears with a 9:5 ratio so that one full step is exactly 1 degree of motion (and with microstepping you can also get 0.5 degree with a half step or lower resolution)

Now that you know how many steps are a degree of rotation you can just configure the stepper motor in the setup menu with this number in the field where you enter the steps for one Millimeter of movement.
Now when you send the command to move "1 mm" it will actually turn 1 degree. So for your setup 1 mm would just be equivalent to 1 degree

pkerspe, 2020-10-14:
Please specify what you mean with "reset the position when..."

forna91, 2020-10-14:
I have a reduct stepper motor so the precision is not a problem,  in my case the stepper turn a "wheel" and a complete turn is 0- deg, if I do one turn and half I wold like to be at 180° and not 540° or mm in the case of this library, can I reset the position after passed the 359.999mm?

pkerspe, 2020-10-14:
You need to answer my question in regards to what you mean by reset, otherwise I cannot help.
Do you just want to rest the value that is displayed in the UI of the stepper motor server or are you using the Rest API or Command line interface and you want to rest the internal counter of the stepper motor server to 0?

closed 2020-10-15 by forna91

pkerspe, 2020-10-15:
what you need is probably kind of an event listener / callback function registration that allows you to reset the internal step counter.
Something like registerTargetPositionReachedCallback() but for position updates.
It is not implemented at the moment, but it should not be to difficult to implement something likes this with only a little decrease in performance.