pkerspe/ESP-FlexyStepper #5: Stepper motor not stepping all steps with this library

xander-m2k, 2021-01-04:
So when I use this library with the following parameters:
```StepsPerRevolution``` : 200
```SpeedInStepsPerSecond``` : 1000 (this corresponds to 500 microseconds in between HIGH and LOW of stepperpin)
```AccelerationInStepsPerSecondPerSecond``` : 2000 (this was my preferred step up and down speed)
```AccelerationInStepsPerSecondPerSecond``` : 2000
I set these values with the corresponding ```set``` functions.

I then use ```stepper.moveToPositionInSteps(200)``` and ```stepper.moveToPositionInSteps(0)``` to move it back and forth, but my stepper does not land at the same position. (So I am just using the Example with some altered values) In fact, it stops earlier than the 200 steps. I don't know what it is, but I would really like to use this library because it has acceleration. 

Let me know what I am doing wrong please, or if your library needs an update. 

FYI, when I do this:
```
for(int i = 0; i < 200; i++){
    digitalWrite(MOTOR_STEP_PIN, HIGH);
    delayMicroseconds(500);
    digitalWrite(MOTOR_STEP_PIN, LOW);
    delayMicroseconds(500);
}
```
It works perfectly fine and lands just on the same spot as before a full rotation.
I also noticed that when I used the default values, my stepper motor just starts grumbling all around, because of the long delay between a HIGH/LOW of the stepperpin. (I am using the "Microstep Driver", just look it up on Google)

So it should be my hardware, or something is wrong in your acceleration, but let me know, I am pretty eager to try a fix.

pkerspe, 2021-01-04:
Thanks you for your report.
Sounds like you are loosing some steps.
I never really checked with an external pulse counter if any step signals are lost, so I cannot for sure say if it is a bug or not.

There are several possible reasons on hardware side that could cause loosing steps:
- to high acceleration or speed-> Try lowering both values to half (or less) and are if you experience the same issues
- to low current setting in your stepper driver (change this setting at your own risk, you might also fry your stepper motor if you exceed its limits)
- to high load on the stepper motor (not sure if this applies for you or you have any load on your stepper motor at all)
- maybe you enabled micro stepping and in combination with other id the above mentioned causes lose steps due to the decrease in torque

Please let me know if any of this helped to improve the situation.
Currently I am not sure about how many steps might be lost in your scenario since you are not so specific about this.

I suggest you also set the driver to full steps and then only issue a single step to the stepper via the library and do that a couple of times, just to see if any steps are lost. Or you try disabling acceleration and see if you also loose steps then.

xander-m2k, 2021-01-04:
1. I tried changing ```periodOfSlowestStep_InUS = 1000000.0 / sqrt(2.0 * acceleration_InStepsPerSecondPerSecond);``` to a bigger value so I removed a 0 from that 1000000, which resolves the problem in losing steps, it now of course does not accelerate as smooth, but at least that is where the problem is.
2. It is definitely not the low current, I set it to 1A and that works like a charm, pretty strong also. My stepper motor is rated for 2,5A but the nice thing about this driver is, that you just have manual switches to switch to a specific current. 
3. I have used the DVR8825 and the A4998 drivers, and just had an awful experience in setting up the right current, they just fry too fast, and it is hard to get right. They also tend to get hot pretty fast. The driver I have now is just big, but has a nice heatsink and is pretty dummy-proof I must say, maybe a recommendation for you!
4. I just discovered that switching to a 1/16 microstep works way better, and is way smoother (but needs more steps of course). Speed is not an issue. Maybe it does bring some hick-up complications later on when adding some more hardware components, but that is for another time. In conclusion, full stepping in combination with acceleration/deceleration just does not work well.

Thanks for the help!

xander-m2k, 2021-01-04:
Also, maybe something to add to your library. 
I am trying to calibrate between two points, so limit switches on both sides. This means I will first be calibrating on the home position (which you do have a function for), and then move to the other side, until the limit switch hits, and setting that as ```MAX_STEPS```, so it cannot move any further than that point. 
That means if you try to do an absolute or relative-move, and that comes above ```MAX_STEPS```, it will just stop at MAX_STEPS.

I would really appreciate it if you would do that for me!

pkerspe, 2021-01-04:
Please open a separate ticket for the feature request, this is a different issue than the problem description you opened this ticket for.

closed 2021-01-04 by xander-m2k