pkerspe/ESP-FlexyStepper #6: New feature: calibrating between two points (home and max position)

xander-m2k, 2021-01-05:
I am trying to calibrate between two points, so limit switches on both sides. This means I will first be calibrating on the home position (which you do have a function for), and then move to the other side, until the limit switch hits, and setting that as ```MAX_STEPS```, so it cannot move any further than that point.
That means if you try to do an absolute or relative-move, and that comes above MAX_STEPS, it will just stop at ```MAX_STEPS```.

You can now use the limit switch feature to get around this nonexistent feature, but the following functions could be implemented:
```void startHomeAndLimitCalibration(int beginLimitSwitch, int EndLimitSwitch)```
(maybe an extra callback function for when this is finished)
```int getMaxSteps()```
```void setMaxSteps(int maxSteps)```
```void moveToMaxSteps()```
```bool isHomeAndLimitCalibrated()```

There could be something I am forgetting, but I think this gets you a good idea of what I mean

FrankNz, 2024-04-07:
Hi Firstly thank you for the work you are doin got this software. We are using it at an Aviation museum to drive a model of a Link Trainer. 
Added to the above..
It would be useful to know when the limit switch (Min || Max) is activated. If I do not know which switch is activated (eg at power on) I could try drive the stepper in the wrong direction which would result in some very bent metal!
Is some work on this problem being contemplated? possible timing?

pkerspe, 2024-04-07:
Hi Frank,

You could call Arduinos digitalRead(...) function to check which limit switch is triggered after booting. Then you need to set the direction of the initial move accordingly to move in the opposite direction.

FrankNz, 2024-04-07:
Hi Paul,
Thanks for the quick response and excellent solution.
Frank