pkerspe/ESP-FlexyStepper #7: [E][esp32-hal-misc.c:94] disableCore0WDT(): Failed to remove Core 0 IDLE task from WDT

Voha888, 2021-01-09:
I have the message in console:
`[E][esp32-hal-misc.c:94] disableCore0WDT(): Failed to remove Core 0 IDLE task from WDT`
when adding more the one stepper service, but all works fine

I think, we need check disabling watchdog in `void ESP_FlexyStepper::startAsService(void)`

pkerspe, 2021-01-09:
Indeed I never checks this use case of starting multiple services.
Feel free to submit a pull request if you have a working solution

Voha888, 2021-01-09:
This is not a critical check. I thought about adding the following functionality, although I'm not sure if it is in demand:
the ability to specify the core on which it will be executed when creating a Task, for example:

`stepper.startAsService ();`  
the task will be created on default core

`stepper.startAsService (0);`
the task will be created on core 0

`stepper.startAsService (1);`
the task will be created on core 1

pkerspe, 2021-01-09:
This is a good idea, I will check on that tomorrow, should be easy to implement

Voha888, 2021-01-10:
If you wish, I can create a pull request with these changes.

pkerspe, 2021-01-10:
Yes, Please go ahead, I will review ASAP once I receive the PR

Thanks

Voha888, 2021-01-13:
#Voha888:adding-dual-core-support

_Due to my poor knowledge of the English language, I did not add comments to the code._
I made two commits.
The first one adds the ability to specify the kernel for the background process
The second, when creating multiple instances of the process (several stepper motors), removes the message from Serial: `[E] [esp32-hal-misc.c: 94] disableCore0WDT (): Failed to remove Core 0 IDLE task from WDT`
It was not done very nicely, but I did not find a method to check if WDT is disabled or not.

pkerspe, 2021-01-14:
thanks for the updates @Voha888, I will review tonight and merge if all is good

Voha888, 2021-01-15:
In the examples, I did not find the use of several stepper motors at the same time.
When you add the changes, I can create an example using multiple stepper motors on multiple processor cores.

pkerspe, 2021-01-15:
Sure can do.
I put some comments on your pull request for minor changes to be performed before merging. Maybe you can have a look at it

Voha888, 2021-01-15:
Thank You!
I see you changed the name of my pull request, but I don't see any other fixes. Please tell me where to go to see them?

pkerspe, 2021-01-15:
I added some comment on some lines of code that you changed in the pull request

Voha888, 2021-01-16:
Sorry, but I don`t see any your comments on this link https://github.com/pkerspe/ESP-FlexyStepper/pull/10/files
Where have you posted these comments? Give me a link please.

pkerspe, 2021-01-16:
Sorry, my bad, I forgot to submit the review.you should see it now

closed 2023-01-21 by pkerspe