pkerspe/ESP-FlexyStepper #8: Adding stepper without DIR pin

Voha888, 2021-01-09:
To save the number of free pins, I needed to implement this, but there is no support in the library.
I could do it if the authors of the project allow

pkerspe, 2021-01-09:
Please feel free to submit a pull request, I will check and merge it if all is ok

Voha888, 2021-01-09:
I created first pull request in my life, i think it"s ok))

https://github.com/pkerspe/ESP-FlexyStepper/pull/9

Description:
I took into account the contents of the file
### core \ packages \ framework-arduinoespressif32 \ cores \ esp32 \ esp32-hal-gpio.c
We'll assign the default DIR pin number 255:
`void connectToPins (byte stepPinNumber, byte directionPinNumber = 255);`

 For pinMode, a check is required because there is a code like this:
`uint32_t rtc_reg = rtc_gpio_desc [pin] .reg;`
where the rtc_gpio_desc array has only 40 cells, and our nonexistent pin is 255, so I added a check:
```
if (directionPin <255) {
pinMode (directionPin, OUTPUT);
}
```

We do not change the digitalWrite function in any way, because the value 255 will not be used in it:
```
extern void IRAM_ATTR __digitalWrite (uint8_t pin, uint8_t val)
{
    if (val) {
        if (pin <32) {
            GPIO.out_w1ts = ((uint32_t) 1 << pin);
        } else if (pin <34) {
            GPIO.out1_w1ts.val = ((uint32_t) 1 << (pin - 32));
        }
    } else {
        if (pin <32) {
            GPIO.out_w1tc = ((uint32_t) 1 << pin);
        } else if (pin <34) {
            GPIO.out1_w1tc.val = ((uint32_t) 1 << (pin - 32));
        }
    }
```

closed 2021-01-10 by Voha888