pkerspe/ESP-FlexyStepper #11: Strange Movement

conaito, 2021-02-10:
I currently try to get my project done with this library (leaved ESP Stepper Motor Server). Looks like working good so far but have a issue on movements of stepper (in loop function). the stepper rotate correct but on reached position he move a little bit back in other direction before stop and reached target position. can you explain what i do wrong? 

#include <Arduino.h>
#include <ESP_FlexyStepper.h>

// IO pin assignments
const int MOTOR_STEP_PIN = 2;
const int MOTOR_DIRECTION_PIN = 15;
const int EMERGENCY_STOP_PIN = 13; //define the IO pin the emergency stop switch is connected to
const int LIMIT_SWITCH_PIN = 12;   //define the IO pin where the limit switches are connected to (switches in series in normally closed setup against ground)

// Speed settings
const int DISTANCE_TO_TRAVEL_IN_STEPS = 2000;
const int SPEED_IN_STEPS_PER_SECOND = 300;
const int ACCELERATION_IN_STEPS_PER_SECOND = 800;
const int DECELERATION_IN_STEPS_PER_SECOND = 800;

// create the stepper motor object
ESP_FlexyStepper stepper;

int previousDirection = 1;

//this function gets called whenever the stepper reaches the target position / ends the current movement
//it is registerd in the line "stepperConfiguration->getFlexyStepper()->registerTargetPositionReachedCallback(targetPositionReachedCallback);"
void targetPositionReachedCallback(long position)
{
  Serial.printf("Stepper reached target position %ld\n", position);

  // previousDirection *= -1;
  // long relativeTargetPosition = DISTANCE_TO_TRAVEL_IN_STEPS * previousDirection;
  // Serial.printf("Moving stepper by %ld steps\n", relativeTargetPosition);
  // stepper.setTargetPositionRelativeInSteps(relativeTargetPosition);
}

void EmergencyStopTriggered()
{
  Serial.printf("Emergency Switch triggered\n");
}

void EmergencyStopReleased()
{
  Serial.printf("Emergency Switch released\n");
}

void ICACHE_RAM_ATTR emergencySwitchHandler(){
  stepper.emergencyStop(true);
  // // we do not realy need to debounce here, since we only want to trigger a stop, no matter what.
  // // So even triggering mutliple times does not realy matter at the end
  // if (digitalRead(EMERGENCY_STOP_PIN) == LOW) // Switch is configured in active low configuration
  // {
  //   // the boolean true in the following command tells the stepper to hold the emergency stop until reaseEmergencyStop() is called explicitly.
  //   // If ommitted or "false" is given, the function call would only stop the current motion and then instanlty would allow for new motion commands to be accepted
  //   stepper.emergencyStop(true);
  // }
  // else
  // {
  //   // release a previously enganed emergency stop when the emergency stop button is released
  //   stepper.releaseEmergencyStop();
  // }
}

void limitSwitchHandler()
{
  // limitSwitchState = digitalRead(LIMIT_SWITCH_PIN);
  // lastDebounceTime = millis();
}

void HomeReachedCallback()
{
  Serial.printf("Stepper is now on Home Position\n");
}

void setup()
{
  Serial.begin(115200);

  //set the pin for the emegrenxy witch to input with inernal pullup
  //the emergency switch is connected in a Active Low configuraiton in this example, meaning the switch connects the input to ground when closed
  pinMode(EMERGENCY_STOP_PIN, INPUT_PULLUP);
  //attach an interrupt to the IO pin of the switch and specify the handler function 
  attachInterrupt(digitalPinToInterrupt(EMERGENCY_STOP_PIN), emergencySwitchHandler, CHANGE);

  //attach an interrupt to the IO pin of the limit switch and specify the handler function
  attachInterrupt(digitalPinToInterrupt(LIMIT_SWITCH_PIN), limitSwitchHandler, CHANGE);

  // connect and configure the stepper motor to its IO pins
  stepper.connectToPins(MOTOR_STEP_PIN, MOTOR_DIRECTION_PIN);
  // set the speed and acceleration rates for the stepper motor
  stepper.setSpeedInStepsPerSecond(SPEED_IN_STEPS_PER_SECOND);
  stepper.setAccelerationInStepsPerSecondPerSecond(ACCELERATION_IN_STEPS_PER_SECOND);
  stepper.setDecelerationInStepsPerSecondPerSecond(DECELERATION_IN_STEPS_PER_SECOND);

  //register the callback function, to get informed whenever the target postion has been reached
  stepper.registerTargetPositionReachedCallback(targetPositionReachedCallback);
  //you can also register for other events using (these accept currently only functions without any parameters):
  stepper.registerEmergencyStopReleasedCallback(EmergencyStopReleased);
  stepper.registerEmergencyStopTriggeredCallback(EmergencyStopTriggered);
  stepper.registerHomeReachedCallback(HomeReachedCallback);
  // stepper.registerLimitReachedCallback(...);

  // Not start the stepper instance as a service in the "background" as a separate task
  // and the OS of the ESP will take care of invoking the processMovement() task regularily so you can do whatever you want in the loop function
  stepper.startAsService();

  //send the stepper on its journey. Once the intial position has been reached, the callback function will get triggered and send the stepper to a new destination
  //stepper.setTargetPositionRelativeInSteps(DISTANCE_TO_TRAVEL_IN_STEPS * previousDirection);

  //stepper.goToLimitAndSetAsHome(HomeReachedCallback);
  stepper.moveToHomeInSteps(previousDirection,300,5000, LIMIT_SWITCH_PIN);
  //stepper.setCurrentPositionAsHomeAndStop();
}

void loop()
{
  // Notice that you can now do whatever you want in the loop function without the need to call processMovement().
  // also you do not have to care if your loop processing times are too long.
  // stepper.moveRelativeInSteps(200);
  // delay(1000);
  Serial.printf("Actual position is %i\n", stepper.getCurrentPositionInSteps());
  delay(5000);
  stepper.moveRelativeInSteps(400);
  delay(5000);
  stepper.moveRelativeInSteps(400);
  delay(5000);
  stepper.moveToPositionInSteps(0);
}

pkerspe, 2021-02-10:
nothing is wrong, this is the intended behaviour as you can see in this part of the code:
https://github.com/pkerspe/ESP-FlexyStepper/blob/40d5fcb1a17c60469e6cf681705519c8321761dc/src/ESP_FlexyStepper.cpp#L915

conaito, 2021-02-10:
oh you may understand me wrong. this i have see and love it but thats not i talking about. i mean in the loop function of my code all positions movements do that

pkerspe, 2021-02-10:
maybe you are moving in the wrong direction and continuously trigger the limit switch?
You set the variable `previousDirection` to 1 and never change it again. So you move to the limit switch in a positive direction.
then in the loop you always call `moveRelativeInSteps`with a positive value as well...the seems strange, maybe it should be a negative value there.
But not sure, just a guess for now since I cannot analyse this fully without a test rig.

conaito, 2021-02-10:
can‘t not really be the point. exact same issue are with one of your original examples

conaito, 2021-02-10:
is there any chance that you soon can look to this?

pkerspe, 2021-02-10:
ok, I tested it and can confirm your observations. Need to look into the possible cause on the weekend

conaito, 2021-02-10:
great!! thank you so much. hope you find the issue

pkerspe, 2021-02-10:
Found the issue: you are starting the stepper library as a service but then only use blocking calls.
Remove the line `stepper.startAsService();`and it will work without the overshooting.
Or you start it as a service and only use the non blocking functions with callbacks.

closed 2021-02-10 by pkerspe