pkerspe/ESP-FlexyStepper #12: Deceleration with position rather than speed.

dcentri, 2021-02-11:
HI,

Thanks for this awesome project. Am having an issue with the `setTargetPositionToStop() ` function. 
I was trying to achieve continuous motion with the library and settled for calling `setTargetPositionRelativeInSteps(CONSTANT)` so I basically extend the relative position whenever the target is almost reached.
I just realized there is two issues with this.

1. When I call `setTargetPositionToStop()` to decelerate to a stop it sometimes takes as much time to stop as It has been running. This, I think is because it uses the position to decelerate instead of the speed.
2. `setTargetPositionRelativeInSteps(CONSTANT)` seems to accumulate the relative position and thus could overflow eventually

Is there a way to fix the `setTargetPositionToStop()` so that it decelerates as per speed not position and is there a better way for continuous motion?

pkerspe, 2021-02-11:
Can you please describe what you actual goal in your project is?
If you only want to spin the motor for an unlimited time at a constant speed, this libaray is totally overkill, you can just setup the PWM module of the ESP to output a PWM signal with the required frequency and you are done.
If you need a specific number of pulses only you can combine with the hardware pulse counter module.
I posted a example code in the Issue #4 (https://github.com/pkerspe/ESP-FlexyStepper/issues/4#issuecomment-774678853)

dcentri, 2021-02-11:
I am working on a paper separation mechanism. Kind of like a printer, picking a paper from a stack but with a lot of variability on paper size and type. I need to coordinate 2 motors as per my design. 

I have limit switches that start and stop the the motors and everything seems to work well, except that when I call the `setTargetPositionToStop()`. The motor seems to keep moving as though it is more corelated with the targetPositionRelativeInSteps  other than the motor speed. I would expect the deceleration to be the same at the same speed but this is not the case.

I had a look at the PWM example and for my application the library has all the required features(background task, variable speed, acceleration/deceleration, easier to manage multiple motors ++).

I hope this clarifies things and a huge thanks for your work!

closed 2022-09-21 by pkerspe