pkerspe/ESP-FlexyStepper #15: How to invert the motor direction?

anggiatm, 2021-04-08:
im new in c++ programing, 
i use this library for dual wheel robot, 
the right wheel is normal direction
but the left wheel is flipped
for moving forward just simply call the function
e.g
`motor_right.setTargetPositionInMillimeters(target);`
`motor_left.setTargetPositionInMillimeters(-target);`
for long code im so confusing in `-` symbol

i think to solve the problem just simply invert the left motor direction, so i can put code in the same target value
e.g
`motor_right.setTargetPositionInMillimeters(target);`
`motor_left.setTargetPositionInMillimeters(target);`

how to invert the direction? thanks
thanks

pkerspe, 2021-04-08:
well, you provided the solution already: you simply invert the value for the second motor, that is the easiest way for sure.
You have three other alternatives I come up with spontaneously:
**1. do it in hardware:**
Add a a signal inverter to the pin that controls direction of the second stepper motor. You can use a simple NOT gate to do this. It will simply convert the high signal to a low signal and the other way round. There are tons of low cost ICs on the market to do this, just one random example is the 74HC14. https://assets.nexperia.com/documents/data-sheet/74HC_HCT14.pdf
You can also with two more parts use an NPN or PNP transistor to do the same.

**2. do it in software by following the path you already went, just create a central function that accepts one parameter and then issues calls to the both setTargetPositionInMillimeters() function and inverting the values. This should be less "confusing" when using it.
```
void setTargetForBothMotors(float distanceToMoveInMillimeters){
    motor_right.setTargetPositionInMillimeters(distanceToMoveInMillimeters);
    motor_left.setTargetPositionInMillimeters(-distanceToMoveInMillimeters);
}
```

**2. do it in software by modifying the library to accept a parameter in the setup routines to define if the direction pin shall be active low or active high, this involves some more changes and is probably not suitable if you are new to programming.
A starting point would be to update the `void ESP_FlexyStepper::connectToPins(byte stepPinNumber, byte directionPinNumber)` function to add a third parameter like  `signed int directionPinActiveState`, then add a new class property with the same name and assign the value to it. Possible values would be -1 and 1.
Then whenever the direction pin state is changed in the code you multiply with the directionPinActiveState property value. Basically this affects 5 places in the whole code that need to be changed, so not the biggest change. All lines with "digitalWrite(directionPin,...)" would need to be adopted basically

closed 2021-04-08 by pkerspe

anggiatm, 2021-04-09:
whoa.. just thought resolved with hardware modification, but i wil not add extra IC
i will swap one coil pair of motor. i found from this [link](https://electronics.stackexchange.com/questions/70643/how-to-reverse-rotation-direction-of-stepper-motor#:~:text=To%20change%20direction%20you%20need,with%20the%20same%20drive%20signal.)
but if the library have invert function that awesome. like `motor_left.setInvertedDirection(boolean invert);`
thanks for suggesting hardware modification.. Sorry for my bad English