pkerspe/ESP-FlexyStepper #16: Spinning and spinning and spinning...

plague-doctor, 2021-04-21:
Hi, your lib is awesome! Thank you very, very much for writing it.
Is there a way to just keep rotating the stepper with a desired speed for very, very long time? Without specifying when to stop? Just keep spinning until the end of time... or a condition is met.

pkerspe, 2021-04-21:
Hi @plague-doctor 
thank you for your feedback.
I think the function you are looking for is generally referred to as "jogging".
Currently the library does not implement this function. You could of course set the target of the movement to an incredibly high number, but it would still come to an end at some point.

Depending on your use case you could probably get away with a much simpler approach than using this library.
If all you need is a constantly spinning motor at a fixed speed while a specific condition/flag is set, you can use a PWM output on your MCU and generate a square wave with a fixed frequency at one of the IO-Pins of the MCU.
You connect this IO PIn to your step-input on your stepper driver.
You use a second IO Pin to to connect to the direction pin of the driver to control the direction (or if you only want it to spin in one direction only, you can directly tie that stepper driver in to VCC or GND (depending on direction your want it to spin).

Of course this option would not implement acceleration / deaccelration.

Now in order to implement the function to only spin when a condition is met, you simply activate/deactivate the PWM output according to your requirements.

Does this help?

plague-doctor, 2021-04-21:
@pkerspe Yes it helps indeed. However when I have finally found your lib, and it allows me to run the motor as a service, I can do whatever "perversion" I want in the main loop. I can monitor and/or update the speed over the network. You have spoiled me with your lib :-) 
So taking the motor for a jogging, and run this as a service, is what I really want. I do not want to loose the acceleration feature. I don't care for deceleration so much. As long as I can start slowly, then speed the revs up, and keep this as long as last proton decays (or motor dies, or whatever) I will be happy,
My project is a very simple one. I build a sophisticated IoT Rocks Tumbler. I would like to be able to run it continuously for 5-10 days and be able to send all interesting statuses to MQTT Server.

pkerspe, 2021-04-22:
ok, if it wasn't for the acceleration feature, the PWM solution would be best for you, since it runs in Hardware and leaves the CPU completely free for other tasks (given that your MCU has a hardware PWM module, which most MCUs have).
But of course out of the box this does not come with an hardware "acceleration function" or slow ramp up on Arduinos.

At the end what you need is based on most LED fading examples you can find on the net, like for example this random one I just googled:
https://techexplorations.com/guides/esp32/begin/pwm/ (Note: you need to change the part that only sets the brightness level by a call to change the PWM frequency rather than the LED brightness, since the stepper driver does not care about the duty cycle, but the frequency, but other than that it should be the same).

Instead of an LED you just connect the stepper driver step-input pin to the GPIO Pin.

LED Control of the ESP32 is described here:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/ledc.html#ledc-api-change-pwm-signal

I can work on a jogging function in the future for this library (or I accept pull request for this feature of course) but currently I have quite a bunch of other side project with higher priority.

Thanks for explaining also the use case of yours, of course for spinning 5-10 days at a constant speed, the solution with just using a large target position will not work, these is no such number in a 32 bit device large enough for that amount of steps I guess (depending on the speed of course, but in general not a good solution)

plague-doctor, 2021-04-22:
Thank you for the explanation. I have found a workaround for huge numbers (I just calculate them of fly in the main loop and always grow).
I will wait for you to find few minutes to implement the jogging function. I wish I could help with a PR but my C++ skills are a bit rusty.

renno-bih, 2022-03-13:
This feature will be very helpful, it will be great to see it in the future.

pkerspe, 2023-06-30:
there is a set of functions `startJogging` and `stopJogging` to achieve the desired outcome.
Yet it is not a true "endless" jogging implementation, it sets a target position of +/- 2.000.000.000 steps and starts going.
I added a new function to reset the internal step counter so you could call `setCurrentPositionInSteps(0)` from time to time to reset the position counter and prevent the step counter to overflow at some point.
but with potentially 2 billion steps before that overflow happens, you should be good for some spinning in a regular use case. E.g. if you want to run the stepper at 20khz it would be fine for jogging for more than a day. In other scenarios you just call the  `setCurrentPositionInSteps(0)` function in your main loop once an hour or so.

pkerspe, 2023-06-30:
new release 1.4.6. contains needed new function

closed 2023-06-30 by pkerspe