pkerspe/ESP-FlexyStepper #19: Sign Ignored in stepper.moveRelativeInSteps

HoinJitGub, 2021-06-15:
Hello

Issue using Example1_BasicStepper.ino

1. Built ESP_FlexyStepper for ESp32_Devkitc_V4 with VSCode, PlatformIO on Ubuntu 18.4, using a TB660 driver. The motor only spins in the same direction with both of the statements stepper.moveRelativeInSteps(200) & stepper.moveRelativeInSteps(-200).

2. Built FlexyStepper for ESp32_Devkitc_V4 with VSCode, PlatformIO on Ubuntu 18.4, using a TB660 driver. The motor only spins in the same direction with both of the statements stepper.moveRelativeInSteps(200) & stepper.moveRelativeInSteps(-200).

3. Built FlexyStepper for Arduino ZERO with VSCode, PlatformIO on Ubuntu 18.4, using a TB660 driver. The motor only spins in the same direction with both of the statements stepper.moveRelativeInSteps(200) & stepper.moveRelativeInSteps(-200).

4. Built FlexyStepper for Arduino UNO with VSCode, PlatformIO on Ubuntu 18.4, using a TB660 driver. The motor spins in opposite directions with  stepper.moveRelativeInSteps(200) & stepper.moveRelativeInSteps(-200).

5. Built FlexyStepper for Arduino UNO & ZERO with Arduino IDE in Windows using a TB660 driver. The motor spins in opposite directions with  stepper.moveRelativeInSteps(200) & stepper.moveRelativeInSteps(-200)  for both the ZERO & UNO.

Has this situation arisen before with ESP_FlexyStepper? Any suggestions on how to remedy it?

pkerspe, 2021-06-15:
Hi,

as for your issued:
1.) I would guess the direction pin is not connected correctly
2.) not sure what the difference is to one
3.) same issue with direction pin
4.) not sure how to eyplain that, other then you setup the GPIO Pins differently or wired it up differently
5.) not sure, I cannot support Arduino and FlexyStepper LIbrary (maintained by another developer)

So what you are saying only combination 4 and 5 are working as intended, right?

I suggest you check your wiring and the documentation of your TB660 driver in regards to the direction PIN.

HoinJitGub, 2021-06-15:
Hello

I thank you for your prompt and detailed response.

Test 2. As ESP-FlexyStepper was derived from FlexyStepper, the latter was tested with an ESP32 to determine whether this issue was solely related to the extensions in ESP-FlexyStepper; Test 2. demonstrates that it is not.

Test 3. demonstrates that the issue is not restricted to the ESP32.

Test 4. demonstrated that the TB6600 was working correctly at 5V logic.

Test 5. demonstrated that the TB6600 was working correctly with the ZERO at 3.3V logic and the UNO at 5V logic whilst employing FlexyStepper, which failed in Test 2., as did ESP32-FlexyStepper in Test 1.

The Arduino board results were included to demonstrate that it is not obviously a voltage or wiring issue. Also a TXS0108E logic level converter was employed without effect with the ESP32. All three boards powered the TB6600 logic at 5V but the logic pins of the ESP32 & ZERO are 3.3V. The TB6600 is a complete driver unit, not just the chip itself, which may in fact be a cheaper Toshiba variant. The internal wiring has not been investigated.

As FlexyStepper had the same issue, the results for UNO & ZERO are potentially informative for ESP-FlexyStepper and these other boards shows that the issue is not board specific, as the same code, excepting pin assignment was used.The ESP32 & ZERO used pins Step=3, Direction =4, UNO used Step=11, Direction =12.

I will try the suggested " directly connect the CW/CCW Pin to 3.3V without any resistor".

pkerspe, 2021-06-15:
I close this issue here, since I do not see this as a problem of the library, but feel free to reply to my response and we can discuss further on this topic

closed 2021-06-15 by pkerspe

pkerspe, 2021-06-15:
please post the link to the datasheet of the exact stepper driver Unit you are using, maybe I can have a look at it.

HoinJitGub, 2021-06-15:
I suggest that you wait until I have applied your test and renewed all of the wiring on Wednesday. I have a logic analyser with pulseView and will use it if the issue persists.

The driver was supplied without documentation. However it appears to be one of many clones, such as https://www.makerguides.com/wp-content/uploads/2019/10/TB6600-Manual.pdf.

HoinJitGub, 2021-06-15:
These tests were Common Anode. I will also try Common Cathode with this library tomorrow. Both were used successfully with previous UNO tests in Arduino IDE.