pkerspe/ESP-FlexyStepper #20: goToLimitAndSetAsHome and limit switch

vitalibr, 2021-06-25:
hi!

I'm trying to understand the code to understand how this function goToLimitAndSetAsHome uses the limit switch if there is no way to inform the limit switch pin.

I was using the moveToHomeInSteps, but goToLimitAndSetAsHome is a non block function (better for me).
Do I need to check the limit switch in my main loop the same way in this example? https://github.com/pkerspe/ESP-FlexyStepper/blob/master/examples/Example4_Limit_and_latching_EmergencySwitch/Example4_Limit_and_latching_EmergencySwitch.ino

vitalibr, 2021-06-25:
Ok, so probably I need to use setLimitSwitchActive in my main loop to trigger the limit switch for the ESP-FlexyStepper

pkerspe, 2023-01-21:
Example [4 limit and latching EmergenccySwitch](https://github.com/pkerspe/ESP-FlexyStepper/tree/master/examples/Example4_Limit_and_latching_EmergencySwitch) shows how to use Limit Switches.

closed 2023-01-21 by pkerspe