pkerspe/ESP-FlexyStepper #21: SetSpeed(0) doesnt stop the motor

dansoskin, 2021-08-29:
hey, id like to use jog functions and change speed on the go, but setting the speed to 0 doesnt stop the motor and it keeps spinning slowly

pkerspe, 2021-08-29:
Did you try setting the distance/target to zero?

pkerspe, 2021-08-29:
Since you also stated you use the jogging function, did you actually call the stopJogging() function?

dansoskin, 2021-08-29:
I want to perform PID on the motor's speed, and I need the motor to stop when the PID output equals to 0

in the library, the setspeed command calculates the interval time by dividing:
desiredPeriod_InUSPerStep = 1000000.0 / desiredSpeed_InStepsPerSecond;
so it divides by zero

maybe theres some kind of quick fix to make the motor stop when the setspeed(0) function is called?
what is the lowest speed possible by the library?

pkerspe, 2021-08-29:
Just call stopJogging function instead of setting speed to zero.
Should be a very simple if/else statement in your code.
As you mentioned already there will be a divide by zero issue when you use an invalid speed value. 
Also the reason for the slow movement you are expecting might be due to rounding issues in the stepping speed calculation routine.
The library simply considers traveling with a speed of zero, as a non valid speed requirement. This to stop the motion there are explicit functions (like stopJogging) that should do the trick

dansoskin, 2021-08-29:
alrighty, thanks for the quick response

closed 2021-08-29 by dansoskin