pkerspe/ESP-FlexyStepper #22: platformio will not compile example code, says CallAsService expects 0 arguments

charlessowers, 2021-11-05:
I cut and pasted the example code in the Readme.md  and added #include <ESP_FlexyStepper.h> but when I try to compile and load it,  Platformio gives me the following error:

> src\main.cpp: In function 'void setup()':
> src\main.cpp:34:27: error: no matching function for call to 'ESP_FlexyStepper::startAsService(int)'
>    stepper.startAsService(1);
>                            ^
> In file included from src\main.cpp:1:0:
> .pio\libdeps\esp32doit-devkit-v1\ESP-FlexyStepper\src/ESP_FlexyStepper.h:60:8: note: candidate: void ESP_FlexyStepper::startAsService()
>    void startAsService(void);
>         ^
> .pio\libdeps\esp32doit-devkit-v1\ESP-FlexyStepper\src/ESP_FlexyStepper.h:60:8: note:   candidate expects 0 arguments, 1 provided
> *** [.pio\build\esp32doit-devkit-v1\src\main.cpp.o] Error 1
> 

If I change the line to "stepper.startAsService();" the program compiles, loads, and runs.

Looking at the source code on GitHub I see that startAsService takes an int for an argument but it looks like for some reason Platformio thinks it takes no arguments. 

Could Platformio be linked to an older version even though it says  "pkerspe/ESP-FlexyStepper@^1.4.4" in the platformio,ini file? I used the library manager to load the ESP-flexyStepper, I am not sure how to do it manually.

pkerspe, 2021-11-05:
Thanks for pointing that our @charlessowers  
I just noticed that the tag name was not consistent (was "1.4.4" instead of "v1.4.4") so I re-published the release.
As for platform.io you can check in the ".pio/libdeps" folder which dependencies have been donwloaded and in which versions.

charlessowers, 2021-11-06:
I uninstalled ESP-FlexyStepper in pio using the library manager and installed it again. I get the same error when trying to run the example from the readme on this git. There is a library.properties file in the .pio/libdeps/ESP-FlexyStepper and line 2 has "version=1.4.4"  same with the library.json file "version": "1.4.4", platformio.ini has "pkerspe/ESP-FlexyStepper@^1.4.4"

pkerspe, 2021-11-06:
Did you check the actual code in the Libdeps folder? 
1.4.4 is the latest version and when I check the tag in GitHub it show the new method signature with the int parameter.
Anyway, I will publish an new release tomorrow to see if that convinces platformio to load the right tag :-)

charlessowers, 2021-11-06:
The source code in the libdeps folder has the startAsService function with zero args
void ESP_FlexyStepper::startAsService(void)

pkerspe, 2021-11-07:
I published a new release (1.4.5) hope that helps. Will most likely take a while for platformio lib manager to pick it uo

closed 2022-03-18 by pkerspe