pkerspe/ESP-FlexyStepper #23: Set speed is slow by about 1.94x

charlessowers, 2021-11-07:
I am using a digital tachometer and am finding that setting the speed is off by about 1.94 times. Is this specific to my board's clock speed? It has been difficult to get specifics on my ESP32 board but I believe my clock speed is 240MHz.

pkerspe, 2021-11-07:
the factor of 1.9 is way to high to be a clock speed issue. I assume you might have micro stepping activated on your driver or maybe you are using acceleration values that are too conservative. All hard to tell without any details on your setup.
I never measured the speed, to be honest, this part has been taken from the original FlexyStepper Library.
Still I believe there is a good chance the problem can be found in your stepper driver configuration and maybe general setup

charlessowers, 2021-11-08:
I haven't tried FlexyStepper but I get the correct speed with AccelStepper. I am driving a Teknik Clearpath SDSK servo motor that takes step and direction input. The driver doesn't microstep but it does have variable steps per revolution. When I set ESP-FlexyStepper to the same steps per revolution as the motor, the speed seems to always be slower by about a factor of 1.9. I am charting various steps per rev settings and speed in steps to try to understand where the problem is. I still suspect there is a math problem somewhere in the library but it could very well be in my code. I'll let you know what I find.

pkerspe, 2022-03-18:
to catch up on this topic here, I connected an Oscilloscope to the ESP and set the stepper on a long journey with 2000 Steps/Second (2khz) and measured the output. The generated signal was between 1.92 and 1.99 khz, so in the expected range (actually a bit low, but that might be due general jitter issue (I tested it by running the webserver in parallel and being connected to the websocket which generates some overhead and jitter).
So I can not measure a step rate that is to high as reported by 1.94 times, the step rate is actually pretty much in the expected range. Closing this issue for now since I cannot reproduce and it might be due to your specific setup / configuration. Feel free to reopen and provide some more details on your exact setup and configuration.

closed 2022-03-18 by pkerspe