pkerspe/ESP-FlexyStepper #24: Cannot get rid of slight shakes WiFi adds

johnsmakar, 2022-02-04:
First of all, Thanks for a great library!

I was able to get to put together a quick test via Serial for testing this lib with my driver:
```
    while (Serial.available() > 0) {
      String command = Serial.readStringUntil('\n');
      if (command.toFloat() < 0) stepper.startJogging(-1);
      if (command.toFloat() > 0) stepper.startJogging(1);
      stepper.setSpeedInRevolutionsPerSecond(abs(command.toFloat()));
      Serial.println(command.toFloat());
    }
```

My next step was to add WiFi functionality to control my stepper motor wirelessly. I notice a tiny bit of vibration has been introduced into the stepper.

I have been messing around changing the core that runs `startAsService()` and trying to find the best priority for `xTaskCreatePinnedToCore(...)` in ESP_FlexyStepper.cpp . I was trying to even add another task to `setup()` rather than using the main `loop()` to have more control. But I cannot seem to find a way to completely remove the added shakes.

Do you have any recommendations on any additional trick I may try to fix this issue? Thanks in advance!

pkerspe, 2022-02-04:
In my opinion you already did all the right steps by changing the core and experimenting with priorities.
In my use cases I was not focussing that much on the vibrations of the motor so I never inspected it all the way. There is another thread though where I pointed out some possible refactoring of the library to move the output signal generation to a hardware module rather than "bit banging".
This is quite some effort though and this I never implemented any of this solutions.

pkerspe, 2022-03-18:
closing since this is a duplicate of #4

closed 2022-03-18 by pkerspe