pkerspe/ESP-FlexyStepper #25: [E][esp32-hal-misc.c:111] disableCore1WDT(): Failed to remove Core 1 IDLE task from WDT

AdrianMolecule, 2022-02-23:
Hi,

I'm trying to use this library to just start a servo and rotate it forever and keep getting this error and some strange repeated movements. Any idea?
Thank you,
Adrian

void setupStepperMotor() {
	Serial.println("Setup Nema ===============");
	pinMode(STEP_PIN, OUTPUT);
	pinMode(DIR_PIN, OUTPUT);
	pinMode(STEPPER_ENABLE_PIN, OUTPUT);
	digitalWrite(STEPPER_ENABLE_PIN, LOW);// the signal is active on LOW
	pinMode(STEPPER_MICROSTEPPING_M0, OUTPUT);
	pinMode(STEPPER_MICROSTEPPING_M1, OUTPUT);
	pinMode(STEPPER_MICROSTEPPING_M2, OUTPUT);
	setMicrostepping(microstepping);
	stepper.connectToPins(STEP_PIN, DIR_PIN);
	stepper.setSpeedInStepsPerSecond(1000);
	stepper.setAccelerationInStepsPerSecondPerSecond(100);
	stepper.startAsService();
	//runStepper();
}
void loop() {
	stepper.setTargetPositionRelativeInRevolutions(10000);
}

pkerspe, 2022-02-23:
I suggest not to call the function without any delay in the loop function.
Your stepper will not spin that fast that you need to call the function that often and without delays.
Try a delay of maybe 1 or more seconds in main loop.
In general if your only goal is to spin the motor continuously without ever stopping, you might want to use the pwm function of the esp directly with your desired frequency and 50% duty cycle instead of using such a library as this one.

pkerspe, 2022-09-21:
Closed due to missing feedback / no updates for 7 months

closed 2022-09-21 by pkerspe