pkerspe/ESP-FlexyStepper #26: How to utilize getCurrentPositionInMillimeters() in flexystepper?

Steve5656, 2022-03-06:
Hi,
I am currently using flexystepper to control a single axis for a plasma cutting application.  It is all working well except I can not figure out the syntax for the getCurrentPositionInMillimeters() function.  I have tried using the following to store the current position in a variable register: CurPos=stepper.getCurrentPositionInMillimeters();.  However this does not seem to work?  Can anyone provide some guidance on exactly what code is required to get this to work?  I am using the command when the stepper is stopped.  Thanks, Steve

pkerspe, 2022-03-07:
Hi @Steve5656, great to hear of use cases where the library is used!
Could you please provide the relevant code snippets of your variable / register declaration and describe what the acutal problem is you are facing?

As a very basic example here an application sample snippet:

#include <ESP_FlexyStepper.h>
const int MOTOR_STEP_PIN = 3;
const int MOTOR_DIRECTION_PIN = 4;
ESP_FlexyStepper stepper;
float positionValue = 0;

void setup() 
{
  Serial.begin(115200);
  stepper.connectToPins(MOTOR_STEP_PIN, MOTOR_DIRECTION_PIN);
  stepper.setStepsPerMillimeter(100); //IMPORTANT, configure according to your stepper configuration how many steps equals one milimeter of movement
}

void loop() 
{
  stepper.setSpeedInStepsPerSecond(100);
  stepper.setAccelerationInStepsPerSecondPerSecond(100);
  stepper.moveRelativeInSteps(200);
  delay(100);
  positionValue = stepper.getCurrentPositionInMillimeters();
  stepper.moveRelativeInSteps(-200);
  delay(1000);
  positionValue = stepper.getCurrentPositionInMillimeters();
}

Steve5656, 2022-03-08:
pkerspe,
Thanks so much for the prompt response.  I ran the code you sent me tonight and it worked fine.  After that I compared my code to what you provided and found out that I had some errors not related to ESP_FlexyStepper that were causing the value of stepper.getCurrentPositionInMillimeters() to actually be zero.  After knowing that stepper.getCurrentPositionInMillimeters() was working properly, I started looking elsewhere and found the bugs causing the problems.  Thanks again for your support!  Regards, Steve