pkerspe/ESP-FlexyStepper #28: [E][esp32-hal-misc.c:94] disableCore0WDT(): Failed to remove Core 0 IDLE task from WDT

maxabba, 2022-03-23:
Hi there, i'm making my own project using your library.
The current setup is: 2 stepper motor with tmc2208 in step/dir confg, the first is set in Rotation and the second is set in Millimeter.

The two big problems is the sync of motion betwen the two stepper and the autio-reset of the esp32 when i sent the 
ESP_FlexyStepper::moveRelativeInRevolutions() comand.

The fist error occurred during the startup is:
`[E][esp32-hal-misc.c:94] disableCore0WDT(): Failed to remove Core 0 IDLE task from WDT`

I set the core on 0 beacuse i use Async_TCP lib which run the task on Core 1
```
#define DEBUG

#define MICROSTEP_X 2                              
#define STEPS_ROTATION_X 50 * MICROSTEP_X

#define MICROSTEP_Y 16
#define STEPS_MM_Y 100 
#define VELOCITY_MM_S_Y 1
#define ACCELERATION_MM_S_S_Y 50

```

```
stepper_x.setStepsPerRevolution(STEPS_ROTATION_X);
stepper_y.setStepsPerMillimeter(STEPS_MM_Y);

  stepper_x.startAsService(0);
  stepper_y.startAsService(0);
```

After that, when i call the up citated function i had this error log:

```
E (33315) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (33315) task_wdt:  - async_tcp (CPU 1)
E (33315) task_wdt: Tasks currently running:
E (33315) task_wdt: CPU 0: FlexyStepper
E (33315) task_wdt: CPU 1: loopTask
E (33315) task_wdt: Aborting.
abort() was called at PC 0x400e4f07 on core 0

Backtrace: 0x4008c438:0x3ffbe170 0x4008c669:0x3ffbe190 0x400e4f07:0x3ffbe1b0 0x40084775:0x3ffbe1d0 0x4000bfed:0x3ffd2b00 0x40089ced:0x3ffd2b10 0x400e5230:0x3ffd2b30 0x400d3483:0x3ffd2b60 0x400d3689:0x3ffd2b80 0x40088b81:0x3ffd2ba0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
```

The calling code is:

```
void handleRotation(AsyncWebServerRequest *request)
{
    float rot = request->arg("rot").toInt();
    float vel = request->arg("rotvel").toInt();
    #ifdef DEBUG
      Serial.println(rot);
      Serial.println(vel);
    #endif 
if (stepper_x.isStartedAsService()){
  digitalWrite(MOTOR_ENNABLE_PIN, LOW);
  delay(100);
    #ifdef DEBUG
      Serial.println("Started as Service and inside handleRotation");
    #endif  
  stepper_x.setSpeedInRevolutionsPerSecond(vel);
  stepper_x.setAccelerationInMillimetersPerSecondPerSecond(50);
  stepper_x.moveRelativeInRevolutions(rot);
  delay(100);
  digitalWrite(MOTOR_ENNABLE_PIN, HIGH);
}
    request->send(200, "text/plane", "1");
}
```

pkerspe, 2023-01-21:
@maxabba I never tested running two instances as service simultaneously to be honest.
I am surprised the watchdog gets triggered since as you can see in the function [bool ESP_FlexyStepper::startAsService(int coreNumber)](https://github.com/pkerspe/ESP-FlexyStepper/blob/41931b9daba9bb03aad00ace6bc97eea3935ee8c/src/ESP_FlexyStepper.cpp#L94) the watchdog should get disabled for the given core number.
Maybe this does not work as intended.
One other issue that might arise: the startAsService() function always uses the same name for the task no matter how often you call it. So maybe the task name should be unique when starting on multiple instances.
I will need to test a bit on that.

pkerspe, 2023-01-21:
Update: the taskname duplication is no problem, the name is, according to the manual, only used for debugging purposes. So the task name is not an issue here.
I also created a new Example [Example7_DualAxis](https://github.com/pkerspe/ESP-FlexyStepper/tree/master/examples/Example7_DualAxis) in the master branch where I created two instances of the stepper class and started both as service on the same core.
I cannot reproduce the kernel panic.

Please also note that your kernel panic is not caused by the ESP Flexy Stepper library according to the provided output:
```
E (33315) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (33315) task_wdt:  - async_tcp (CPU 1)
```
In your case the async_tcp tasks causes the issue.
Maybe you have a long running request handler in there that cause the Watchdog to start "barking" :-)

The message `[E][esp32-hal-misc.c:94] disableCore0WDT(): Failed to remove Core 0 IDLE task from WDT` can be ignored in my opinion. This is triggered since the service is started two times and both intstances try to disable the WDT in the `startAsService` function. But since it is already stopped after the first call the second attempt "fails".
I tried to only conditional disable the WDT with something like this, but it didn't work:

```
TaskHandle_t idle_0 = xTaskGetIdleTaskHandleForCPU(0);
    if (idle_0 != NULL)
    {
      disableCore0WDT(); // we have to disable the Watchdog timer to prevent it from rebooting the ESP all the time another option would be to add a vTaskDelay but it would slow down the stepper
    }
```

pkerspe, 2023-01-21:
for now I have to close the issue since it seems to be caused by another library (namely the async_tcp task).
**You can try to call `disableCore1WDT();` in your code to disabled the watchdog timer for core 1 as well.**

closed 2023-01-21 by pkerspe