pkerspe/ESP-FlexyStepper #30: Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Freeejack, 2022-09-21:
Hi!

Yesterday i run into the problem below:

08:33:30.981 -> Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
08:33:31.028 -> Core 1 register dump:
08:33:31.028 -> PC      : 0x000005dc  PS      : 0x00060030  A0      : 0x800d14d8  A1      : 0x3ffb8be0  
08:33:31.028 -> A2      : 0x3ffbfe78  A3      : 0x000005dc  A4      : 0x00000001  A5      : 0xfffffdb3  
08:33:31.028 -> A6      : 0x00000003  A7      : 0x00060f23  A8      : 0x800d14a9  A9      : 0x3ffb8bc0  
08:33:31.028 -> A10     : 0xfffffdb2  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001  
08:33:31.028 -> A14     : 0x00060020  A15     : 0x00000000  SAR     : 0x00000005  EXCCAUSE: 0x00000014  
08:33:31.028 -> EXCVADDR: 0x000005dc  LBEG    : 0x400029ac  LEND    : 0x400029cb  LCOUNT  : 0x00000000  
08:33:31.082 -> 
08:33:31.082 -> ELF file SHA256: 0000000000000000

I downloaded the Exception Decoder and here are the results:

PC: 0x000005dc
EXCVADDR: 0x000005dc

Decoding stack results
0x400d14d5: ESP_FlexyStepper::taskRunner(void*) at D:\PGRDrive\Dolly\Arduino_dolly_ESP32_Flexy_SUMD/ESP_FlexyStepper.cpp line 127
0x40086161: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

I don't know what the core of the error, i see that the line 127 is the "stepperRef->processMovement();" line in Flexystepper.cpp.
I want to use the Flexystepper library with servos that driven by received data from a radio receiver. The receiver communicate with the ESP32 via the RX pin. So i tried to separate the serial read and the servo code to seprate process/core. The error is the same with core 0 and core 1, i tried both of them.

Can You please help me to find a solution or the mistake i made?

Thnaks!

pkerspe, 2022-09-21:
It is hard to tell why you receive this error, since I do not know about your other code.

Have a look at this page to get more details about the guru mediation error:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#guru-meditation-errors

Also please provide details on the esp32 type you are using, the espressif framework version and also if you get the same error when you use the flex's stepper library examples without any of your code.

pkerspe, 2023-01-21:
will close due to missing feedback to reproduce error

closed 2023-01-21 by pkerspe