pkerspe/ESP-FlexyStepper #31: Setting disallowedDirection doesn't work properly

KDR3bel, 2023-01-21:
Hi,

I'm currently working on an implementation using one stepper motor and two limit switches (separate inputs) on an ESP-32-CAM-module. In the process I noticed that one of my limit switches wasn't working correctly. After checking my own implementation multiple times, I examined the processMovement() function and I think I found a small mistake in the following section (Line 1144):

https://github.com/pkerspe/ESP-FlexyStepper/blob/c9560fcf85e08a789b58c89344ac3a5ecf8153c1/src/ESP_FlexyStepper.cpp#L1141-L1165

When the limit switch ist triggered, I'm calling:  myStepper.setLimitSwitchActive(-1);

https://github.com/pkerspe/ESP-FlexyStepper/blob/c9560fcf85e08a789b58c89344ac3a5ecf8153c1/src/ESP_FlexyStepper.cpp#L213-L224

Corresponding to the definition:

https://github.com/pkerspe/ESP-FlexyStepper/blob/c9560fcf85e08a789b58c89344ac3a5ecf8153c1/src/ESP_FlexyStepper.h#L151

In line 217 the passed byte value is saved in the variable "this->activeLimitSwitch" of type signed char.

https://github.com/pkerspe/ESP-FlexyStepper/blob/c9560fcf85e08a789b58c89344ac3a5ecf8153c1/src/ESP_FlexyStepper.h#L208

I'm not sure, if the following explanation is right, so please correct me, if I'm wrong.

byte can hold numbers between 0 and 255 whereas signed char can hold numbers between -128 and 127.
The passed value of byte = -1 would ussually overflow to a value of 255, but signed char can save it as a value of -1 (see example).

` byte a = -1;

Serial.println(a); //result: 255
signed char b =a;
Serial.println(b); //result: -1
bool c = (a == b);
Serial.println(c);  //result: 0
bool d = ((byte)a ==b);
Serial.println(d); //result: 1`

This leads me to the issue in Line 1144 :

https://github.com/pkerspe/ESP-FlexyStepper/blob/c9560fcf85e08a789b58c89344ac3a5ecf8153c1/src/ESP_FlexyStepper.cpp#L1144

In the comparison this->activeLimitSwitch has a value of -1 and this->LIMIT_SWITCH_BEGIN has a value of 255. Therefore the condition cannot become true, leading to the limit switch not preventing movement in a certain direction. This behaviour is not noticable in the examples, since the limit switches are connected in series  and use the condition in line 1152.

`if ((byte)this->activeLimitSwitch == this->LIMIT_SWITCH_BEGIN)`

Casting this->activeLimitSwitch to byte in the comparison seems to fix the issues in my case. 
Maybe someone could test this potential fix and see, if it works without causing any problems.

pkerspe, 2023-01-21:
Thanks for opening the issue @KDR3bel !
I will check that today and try to provide a fix.
For now I guess the type of the function parameter should be changed from byte to signed char, casting the parameter to byte might also work, but at the end the problem lies in the function accepting the wrong type IMHO.
Will update you once fixed

pkerspe, 2023-01-21:
@KDR3bel I added a new release that should hopefully fix the issue. Can you please test?
https://github.com/pkerspe/ESP-FlexyStepper/releases/tag/v1.4.6

KDR3bel, 2023-01-21:
@pkerspe 
I testet your new release, but it does not solve the issue yet. The condition in the following line still cannot become true, since you didn't change the datatype of LIMIT_SWITCH_BEGIN (byte at the moment).

https://github.com/pkerspe/ESP-FlexyStepper/blob/fbbf032ab3d3e7f576e5c486549b6a2d10cf6104/src/ESP_FlexyStepper.cpp#L1147

Therefore there is now an additional error in the condition in line 218:

https://github.com/pkerspe/ESP-FlexyStepper/blob/c9560fcf85e08a789b58c89344ac3a5ecf8153c1/src/ESP_FlexyStepper.cpp#L213-L224

This condition is now unable to become true aswell, if you pass a signed char calue of -1.

However, changing the dataypes of:

https://github.com/pkerspe/ESP-FlexyStepper/blob/fbbf032ab3d3e7f576e5c486549b6a2d10cf6104/src/ESP_FlexyStepper.h#L151-L152

to:
`static const signed char LIMIT_SWITCH_BEGIN = -1;
  static const signed char LIMIT_SWITCH_END = 1;`

should solve both issues using your new release. 
My implementation doesn't make use of the brake-functionality, so I can't say, if your changes work for these functions.

PS:
I think additional get-functions for the private class attributes would be nice to have. At the moment I'm testing the library  by creating a local copy and declaring all attributes as public, so I can read them out. Maybe there are other ways to do so, but I think these functions might at least come in handy for debugging purposes .

pkerspe, 2023-01-21:
@KDR3bel  fix pushed to main

closed 2023-01-21 by pkerspe

KDR3bel, 2023-01-22:
@pkerspe Thanks for the quick implementations! 😃 👍
The code works fine now. But when I reinstall version 1.4.6 via the Arduino IDE, it does not include the final fix. No clue if it only updates once per version number. A new version number may fix this.🤔 Anyway, installing manually works fine.

commit b78046f1 (2023-01-21): fix #31 variable type