pkerspe/ESP-FlexyStepper #32: moveToPositionInSteps() in constant speed option

christophepersoz, 2023-06-10:
Hi,

You did a really good job with this library, thanks a lot for that.
I'm looking a way to adapt your lib to add a moveToPositionInSteps() in constant speed skipping the calculation of acceleration en deceleration. Unfortunately, I'm maybe doing something wrong as my change do not work and result in a non moving motor.

Is there a way to get that really quickly? Maybe I missed something somewhere in your code that would allows me to do so.

Thanks in advance!
Cheers,

pkerspe, 2023-06-10:
Hi Christophe,
Can you post your code so I can have a look at it?

Kind regards
Paul

pkerspe, 2023-06-10:
Btw you could also just try to set the acceleration values to some rely high numbers which would effectively result in a almost instant speed change

christophepersoz, 2023-06-10:
Hello Paul,

thanks a lot for your reply!

I’m using you lib in a really specific use which isn’t the primary use of it I have to admit - sorry 🤩
I’m feeding the core1 with speed and position values and assign them on core 0 which is in charge of the motor moves only. 
The acc/dec is managed by the speed ramp values provided so it will prevent brutal acc/dec and skipping steps. 
I noticed that removing all the unnecessary part in you lib that implies test and CPU cycles like brakes, limits etc speed up and improve the movement when I’m feeding the core with new speed/pos values each 10 to 20ms. On that way I thought that skipping the calculation of the acc/dec will make sense and will contribute to a better responsiveness of the system; also the overall movement will be much constant as it won’t be acc/dec at each new position to move on. 

I do not know if that’s really clear but to be shorter, that’s why I intend to stick to the minimum calculation and get a faster response of the whole. 

For the moment I was concentrated on the part in charge of the step calculation. 

Thanks a lot!
Christophe

pkerspe, 2023-06-10:
well, it is a bit difficult for me to decide which functions you want to keep from the library in your use case.
The thing is: if you only need to generate a specific amount of step signals and a specific frequency, you might be much better offer simply creating your own logic to simply generate a Squarewave signal for your steppe motor.
Easiest would be to simply write a while loop to set a pin low and high with delays. Obviously not very elegant though
More advanced: use a time and interrupts to generate the signal and count the steps
best: use a hardware module of the ESP32 to generate the signals, I wrote a long text about the different options in one of issue #4  https://github.com/pkerspe/ESP-FlexyStepper/issues/4#issuecomment-774678853

Of course you could still create your own fork of the library and adapt the [processMovement function](https://github.com/pkerspe/ESP-FlexyStepper/blob/master/src/ESP_FlexyStepper.cpp#L1099) to your needs, e.g. you can throw out all the emergency break stuff if you do not need it, also you can remove the break logic, if you do not need it and you can remove limit switch checks etc. The thing is: I do not know what you need of these features, so I cannot provide a solution.

It would be best if you post your code you have so far so we can check

aaryanmurgunde, 2023-08-29:
Hi, @pkerspe 

I have a similar case scenario, what I want to achieve is to do a trajectory execution for a robotic arm. In that case what I receive on the esp32 end is individual point by point for all the joints, having values (position, velocity, acceleration). 
I tried using your lib but since the acceleration and deceleration both happen for a particular target, I get a discontinued motion between points.

The velocity profile and the acceleration are manged by the higher level controller when generating the trajectory, Ideally what I would like to achieve is that I set these values and the motor runs with them and after position is achieved the motor comes to a sudden stop. What this does is as I have a profile build and a next point queued, The execution wont be discontinued. 

If you can tell me what the necessary changes would be for the same that would be Great!

Thank You.