pkerspe/ESP-FlexyStepper #33: Enhancement: Open drain output option

PheebeUK, 2023-06-29:
Currently the driver configures the ESP outputs in push-pull mode, which is great for many use cases. However some stepper drivers operate with an opto-isolated input. These are often best driven with an open drain output. This allows the use of the GPIO pin to be used to switch pull 5v loads to ground, providing sufficient headroom to drive opto-isolators reliably. 

Basically this just means the output pin is either floating or connected to ground. In this mode the ESP32 can sink current on a 5v tolerant pin, which it can't do in a push-pull mode. 

For step and direction, adding a separate call to configure the output type would make sense without any API breaking changes.  The open drain output mode can be set with:
`pinMode(stepPin OUTPUT_OPEN_DRAIN);`

For the brake pin, I'm not sure the best way.

PheebeUK, 2023-06-30:
Yes, that'll work too! Thanks!

pkerspe, 2023-06-30:
@PheebeUK I published a new release v1.4.8 with the requested feature, please test and report back

closed 2023-06-30 by pkerspe

PheebeUK, 2023-07-05:
Thanks @pkerspe - Sorry I haven't had an opportunity to test yet. I hope to crowbar a little time in this weekend.