pkerspe/ESP-FlexyStepper #34: Non linear stepping speeds

JamesF8, 2023-08-09:
Hello Paul,

I've used your stepper library in the past and it's been very useful and a pleasure to work with.
Just started a new project and I'm running into a problem when trying to run a stepper continuously at certain set speeds. It appears that under lower step speeds, I get the correct expected linear speeds, but at higher input speeds the rpm's stay the same even when higher step values are entered. Then at even higher speeds the rpm's shoot up and very non linear speed behavior is observed.

Here's a snipped of the control speed function in my sketch,

   long revolutions = 1000000;   
   int runMotor1 = (revolutions * 6030) * -1;
   int runMotor2 = (revolutions * 800) * rotateDirection;
   
   // Motor at 1 revolution per minute is 6030/60 = 100 steps per second
   int rpmM1 = 100;
   int motor1Diameter = 203;
   float CircumferenceA = (3.1415 * motor1Diameter) / 25.4;
   int Motor1Speed = (rpmM1 * inputSpeed * CircumferenceA) / 60;
   int SPEED_IN_STEPS_PER_SECOND_M1 = (Motor1Speed);
   
   // Motor2 at 1 revolution per minute is 800/60 = 14 steps per second
   int rpmM2 = 14;
   int motor2Diameter= 222.25;
   float CircumferenceB = (3.1415 * motor2Diameter) / 25.4;
   int Motor2Speed = (rpmM2 * inputSpeed * CircumferenceB) / 60;
   int SPEED_IN_STEPS_PER_SECOND = (Motor2Speed);

   Motor1.setAccelerationInStepsPerSecondPerSecond(AccelSaved * 400);
   Motor1.setDecelerationInStepsPerSecondPerSecond(DecelSaved * 400);
   Motor2.setAccelerationInStepsPerSecondPerSecond(Accel2Saved * 400);
   Motor2.setDecelerationInStepsPerSecondPerSecond(Decel2Saved * 400);

   Motor1.setSpeedInStepsPerSecond(SPEED_IN_STEPS_PER_SECOND_M1);
   Motor2.setSpeedInStepsPerSecond(SPEED_IN_STEPS_PER_SECOND);
   Motor1.setTargetPositionRelativeInSteps(runMotor1);
   Motor2.setTargetPositionRelativeInSteps(runMotor2);

The inputSpeed variable ranges from 1 to 60. Motors run at correct speeds up to around inputSpeed of 10, but after that it can vary quite a bit from the correct set speed. Both stepper drivers are set to 800 microsteps. Any help on this would be greatly appreciated.

Thanks in advance!

James

kleurbleur, 2023-11-17:
We also observe stuttering at high speeds. Our hybrid servo (integrated stepper driver) is set at 40000 micro steps and at speeds of 200k stuttering occurs. For now we have flexystepper as a service but we're going to do tests with the processMovement() in the main loop.

pkerspe, 2023-11-17:
Hi James,

I see you are using a float variable (CircumferenceB) in your calculation and then assign to an int value. 
This way you will certainly loose precision. Not sure though if it is the sole reason for the problem.
Can you please explain how much (in percentage) deviation from the expected RPMs you are talking about?

Also are you talking about variation as in "I expect 1000 RPM but measure constant 1100 RPM" or are you talking about jitter (as in non constant RPM values)

pkerspe, 2023-11-17:
Hi @kleurbleur 
I feel you are describing a different issue and also 200k steps per second (or microsteps) at 40000 microsteps (what an insane high number, never worked with a stepper driver that offers such precision...).
Keep in mind that microstepping usually also comes with torque reduction, going at such high speeds and depending on the acceleration values you might as well run into the barriers of physics with you stepper. Best way to find out if it is a software problem or hardware problem is to connect an oscilloscope to the step pin and check if the frequency is stable at your desired 200khz.
Anyhow you might be describing a different problem than James, not sure. Please check the general thread / issue on jitter!