pkerspe/ESP-FlexyStepper #36: bluetooth stepper

arlimbad222, 2023-12-28:
Hello everyone 

I just want to run stepper motor continuous rotation and i am using jogging function i also wnat to change speed via command i am not coading guy but i generate some code can anyone help me with code .

application: 

F-500 to rotate motor at 500rpm (cw)
R-500 to rotate motor at 500rpm (ccw)
S - for stop jogging 

i share my code if any one can help me please help me 

---------------------------------------------------------------------------------------------------------------------------------------------------

#include <ESP_FlexyStepper.h>
#include "BluetoothSerial.h"

const int MOTOR_STEP_PIN = 23;
const int MOTOR_DIRECTION_PIN = 22;

ESP_FlexyStepper stepper;

BluetoothSerial SerialBT;

bool jogging = false;
int previousDirection = 1;  // Initial direction

void setup() {
  Serial.begin(115200);
  stepper.connectToPins(MOTOR_STEP_PIN, MOTOR_DIRECTION_PIN);

  // Initialize Bluetooth Serial with a custom name
  SerialBT.begin("CustomBluetoothName");

  // Uncomment the line below if you want to change the Bluetooth device name programmatically
  // esp_bt_device_name = "CustomBluetoothName";
}

void loop() {
  // Your main loop code goes here

  // Example: Read data from Bluetooth
  if (SerialBT.available()) {
    char command = SerialBT.read();

    // Process the received command as needed
    processBluetoothCommand(command);
  }
  
  // Additional loop logic, if any
}

void processBluetoothCommand(char command) {
  switch (command) {
    case 'F': // Start jogging forward
      startJogging(1);
      break;
    case 'R': // Toggle direction
      previousDirection *= -1;  // Toggle direction
      digitalWrite(MOTOR_DIRECTION_PIN, previousDirection > 0);
      break;
    case 'S': // Stop jogging
      stopJogging();
      break;
    // Add more cases for other commands if needed
  }
}

void startJogging(signed char direction) {
  // Set the speed and acceleration rates for jogging
  stepper.setSpeedInStepsPerSecond(200);  // Adjust speed as needed
  stepper.setAccelerationInStepsPerSecondPerSecond(200);  // Adjust acceleration as needed

  // Start jogging in the specified direction
  stepper.startJogging(direction);

  jogging = true;
}

void stopJogging() {
  // Stop jogging
  stepper.stopJogging();

  jogging = false;
}

arlimbad222, 2023-12-28:
i am taking help of Ai and i think i am near to solve my code here is my second trial code

-----------------------------------------------------------------------------------------------------------------------------------------------

#include <ESP_FlexyStepper.h>

// IO pin assignments
const int MOTOR_STEP_PIN = 22;
const int MOTOR_DIRECTION_PIN = 23;

// Serial command characters
const char FORWARD_COMMAND = 'F';
const char REVERSE_COMMAND = 'R';
const char STOP_COMMAND = 'S';
const char SPEED_COMMAND = 'P';  // Command to change speed

// create the stepper motor object
ESP_FlexyStepper stepper;

// Speed and acceleration settings
int currentSpeed = 100;  // Initial speed in steps per second
const int ACCELERATION_IN_STEPS_PER_SECOND_PER_SECOND = 7000;
const int DECELERATION_IN_STEPS_PER_SECOND_PER_SECOND = 7000;

void setup() {
  Serial.begin(115200);  // USB Serial

  // connect and configure the stepper motor to its IO pins
  stepper.connectToPins(MOTOR_STEP_PIN, MOTOR_DIRECTION_PIN);

  // set the initial speed, acceleration, and deceleration rates for the stepper motor
  stepper.setSpeedInStepsPerSecond(currentSpeed);
  stepper.setAccelerationInStepsPerSecondPerSecond(ACCELERATION_IN_STEPS_PER_SECOND_PER_SECOND);
  stepper.setDecelerationInStepsPerSecondPerSecond(DECELERATION_IN_STEPS_PER_SECOND_PER_SECOND);
}

void startJogging(signed char direction) {
  // Start jogging (continuous movement) in the specified direction
  stepper.startJogging(direction);
}

void stopJogging() {
  // Stop jogging
  stepper.stopJogging();
}

void setSpeed(int newSpeed) {
  // Set the new speed for the stepper motor
  currentSpeed = newSpeed;
  stepper.setSpeedInStepsPerSecond(currentSpeed);
}

void loop() {
  // Handle USB Serial commands
  if (Serial.available() > 0) {
    char command = Serial.read();

    switch (command) {
      case FORWARD_COMMAND:
        // Start jogging forward
        startJogging(1);
        break;

      case REVERSE_COMMAND:
        // Change the direction and start jogging reverse
        stepper.setDirectionToHome(-1);  // Reverse direction
        startJogging(-1);
        break;

      case STOP_COMMAND:
        // Stop jogging
        stopJogging();
        break;

      case SPEED_COMMAND:
        // Change speed based on the value received
        if (Serial.available() > 0) {
          int newSpeed = Serial.parseInt();
          setSpeed(newSpeed);
        }
        break;

      default:
        // Invalid command
        Serial.println("Invalid Command");
        break;
    }
  }

  // Continue processing stepper movements
  stepper.processMovement();
}

pkerspe, 2023-12-28:
Please describe your problem.

I doubt this is about an issue with the Library itself you are talking about, but more a lack of experience in coding.
As long as you cannot describe your problem, I cannot help and in general a forum like stack overflow or the Arduino Forum would be probably better to help you with novice coding questions.

arlimbad222, 2023-12-28:
Okay thanks i will post on that website.

closed 2023-12-28 by arlimbad222