pkerspe/ESP-FlexyStepper #41: Port for STM32 with HAL library

jafarniknam, 2024-02-17:
Hi . I used ESP-Flexy stepper for my position control projects  ( Medical Pumps and & CNC axis and etc )which my MCU was ESP32 and I usually designed GUI with LVGL and of course every thing works fine and perfect .
Now I have a project with STM32 and I'm using STM32Cube IDE with Hal library + TouchGFX + FreeRTOS . so I must modify ESP-FlexyStepper library to work with HAL . firstly I edited all bellow functions : 
digitalWrite() -> HAL_GPIO_WRITE_PIN();
digitalRead() -> HAL_GPIO_READ_PIN();
micros() -> __HAL_TIM_GET_COUNTER(&htim2); ---->  timer 2 configured at 1Mhz clock
delayMicrosecond() -> delay_us(uint16_t us); which is defined as bellow (Not reseting tim2 counter):

`void delay_us(uint16_t us) {`
`	uint32_t startUs = __HAL_TIM_GET_COUNTER(&htim2);`
`	while (__HAL_TIM_GET_COUNTER(&htim2) - startUs < us){`
`		;  // wait for the counter to reach the us input in the parameter`
`   }`
`}`

secondly in my main program , created a task to call processMovement() :
[image]
[image]

In terms of my driver (Lead shine CS-D1008) , minimum pulse width is 2.5 us , I added a delay_us(3) after digitalWrite(stepPin, HIGH) in processMovement() function . 
[image]

In my last test , position control works fine due to accuracy but motor is running with vibration ( not such motion quality and silence motion with ESP32) and I think it's a problem with  timing calculations in this functions : 

`// figure out how long before the next step`
`	DeterminePeriodOfNextStep();`

pkerspe, 2024-02-17:
@jafarniknam I am sorry, but I cannot help you with you Port, since I do not own and have no experience with STM MCUs.

jafarniknam, 2024-02-17:
> @jafarniknam I am sorry, but I cannot help you with you Port, since I do not own and have no experience with STM MCUs.

You helped me before with your excellent library 🥇

closed 2024-07-14 by pkerspe