pkerspe/ESP-FlexyStepper #45: Feature Request - Setting for priority and addition to readme

moose-on-road, 2024-04-28:
Hi, something that would be helpful to amateurs like myself, could you implement a "stepper.setTaskPriority" along the lines of how core selection was added?  

i found bumping this up solved my jittering issues, and reading through some discussion it has helped several other folks. If you know what youre doing, it only takes 2 seconds to find and change the setting, but it took a lot of reading to figure out that changing the setting would be helpful, so maybe a row could be added to the decision matrix.

pkerspe, 2024-04-28:
@moose-on-road can you provide some input for the readme please since you did all the reading already?

moose-on-road, 2024-04-28:
If there was an option to set the priority, the decision matrix could include another row:
Decision matrix:
Scenario :
           appropriate core selected, jittering still present
Suggestion:
          change setTaskPriority() to 2