pkerspe/ESP-FlexyStepper #47: getCurrentPositionInSteps not working when motor is moving

kaito83, 2024-07-14:
I need the actual motor position when motor is moving, but if motor stopped I get the position, but when moves i didnt get any result.

From my stepper layer:
`  rotator.connectToPins(rotator_stp, rotator_dir);
  rotator.setSpeedInStepsPerSecond(speed);
  rotator.setAccelerationInStepsPerSecondPerSecond(accel);
  rotator.setDecelerationInStepsPerSecondPerSecond(accel);
  rotator.startAsService(1);`

`long rot_position() {
  return rotator.getCurrentPositionInSteps();;
}`

And I use moveToPositionInSteps(p);

Or I doing something wrong?

pkerspe, 2024-07-14:
Please post complete code, if you are calling a blocking function to start the movement the behaviour you describe is to be expected and normal.
Don't use blocking functions when using as a service.

kaito83, 2024-07-14:
Thanks, I see know but could you update the description for the getCurrentPositionInSteps but for me a little bit misleading :)

pkerspe, 2024-07-14:
The documentation of moveToPositionInSteps is pretty clear about it being a blocking call.
"Blocking call: start movement in the given direction with a maximum number of steps or until the IO Pin defined by homeSwitchPin is going low (active low switch is required, since the library will configure this pin as input with internal pull-up in the current version). This is a blocking function, **it will not return before the final position has been reached.**"
So no need to update another function documentation if the cause is that you called a blocking function. It is basic understanding of how to develop embedded firmwares and what blocking functions calls are.

closed 2024-07-14 by pkerspe