pkerspe/ESP-FlexyStepper #48: Add support for angles, shortest path.

DanielCardiel, 2024-11-03:
Add support for angles.

In the current iteration, if you move for example from rev 0.9 to .01, instead of moving the shortest part, it will go back doing the long direction.

now if I go from 259 to 1°, given the following code:

`void StepperControl::moveStepperToAngle(float angle)
{
  // Ensure angle is within 0-360 range for safety
  angle = fmod(angle, 360.0);
  if (angle < 0)
  {
    angle += 360.0;
  }
  long steps = static_cast<long>((angle / 360.0) * STEPPER_STEPS_REV);

  Serial.println("Moving to: " + String(angle) + "°");
  Serial.println("Moving to position: " + String(steps));

  // Move to the calculated position in steps
  stepper.moveToPositionInSteps(steps);
}`

It will go back on the long direction not the shortest path.

pkerspe, 2024-11-04:
Please Feel free to create a pull request, then I will review it.
In general so far the library was more intended for linear movements but I do not see a reason to add such functions as long as they do not impact performance. 
For the blocking function set it should not make much difference, for the non blocking part this feature might be a bit more complex / involve more changes.

DanielCardiel, 2024-11-06:
Will do,
I created the helper function, but I mean, if the library can have it from scratch. Would be nice :)

Will do later when I go back to my project :)