pkerspe/ESP-FlexyStepper #50: Is directionPinNumber optional?

Haimana, 2024-12-06:
Hello, 
I need to control a stepper in a single direction, so no need the directionPinNumber
If only stepPin used, like this:

`stepper.connectToPins(stepPin);
`
I get this message in serial:

`E (225545) gpio: gpio_set_level(227): GPIO output gpio_num error
`

Of course, I can define a directionPin but I'm fully out of available GPIOs because a TFT display with parallel connection which will be difficult to replace for other reasons.
So, not a big deal with that message but the question still remain: will work like this without other issues?

Thank you,
Seb

pkerspe, 2024-12-06:
Can you please post all the code including the initialisation of the variables you use to configure the library?
I am not sure how to reproduce the message you get.
In general it should be doable to easily modify the library to ignore directions. I will check once you posted the whole init / setup routine code part

pkerspe, 2024-12-06:
I had a Quick Look in general into the code and it is fine to. It use a direction pin, default value for this pin will be 255 then, and the library will not set the pinmode in that case.
Not sure though about the error message you posted, when exactly is this one showing?

Haimana, 2024-12-06:
thank you for instant reply
I just placed a scope on the stepPin and it seems there are some more issues by my side so I have a little bit more to dig
I'll report back ASAP

koombot, 2025-02-10:
Not sure if it will be helpful, but is it possible to change the pins you are using?  The only time I've seen that error is if the pin being used to output is an input only pin.

Looking at the pinout for that board they shouldn't be, but maybe there is something weird going on if it is a really old board.